elhep/FMC_ADC_128CH_10B #3: Unconnected signal

aw-pl, 2024-03-12:
Signal PDN_GLB not connected to FMC connector.
Not necessary to have design working (checked), but ensure the proper pull-up/pull-down exists in not connecting to FMC

aw-pl, 2025-03-20:
I will double check the Pull-down pin in datasheet to be sure.

We will use this unused port from FMC connector and (if necessary through IC6) to supply it to AFE as calibration pulse (more on AFE issues) from FPGA.

filipswit, 2025-03-20:
The reason PDN_GBL wasn't connected to FMC was because there was no free signals, so we still don't have them.

aw-pl, 2025-03-24:
Reply in #9

aw-pl, 2025-03-24:
> I think there is 14k5 pull-down inside the chip.
> 
>     * Untied not connected signals from translator and tied input (IC6-7) to GND
> 
> 
> [image]
> 
>     * Removed unused signals from SPI_translators sheet
> 
>     * Added external pull-down R5
> 
> 
> [image]

I have checked in datasheet - yes, there is an internal 14.5kOhm pulldown.