elhep/FMC_ADC_128CH_10B #5: Add buffering of ADC IN-

gkasprow, 2025-03-04:
- [ ] add reference + buffering of ADC negative terminal to the adjustable value of 1.1...1.6V 
- [ ] add digital trim pot

filipswit, 2025-03-18:
Shouldn't it be done on AFE board?
You'd like to make sth like this on each IN_N[0..3]?

[image]

gkasprow, 2025-03-19:
One buffer for all 128 channels

gkasprow, 2025-03-19:
exactly, use this ciruit

aw-pl, 2025-03-19:
@gkasprow I think we were also talking to add some simple ADC for monitoring power supply voltages for FMC board (including IN_N readout). Do you confirm to also add this?

filipswit, 2025-03-23:
Since we generate IN_N on FMC, I remove protection diode D9

[image]

gkasprow, 2025-03-24:
right

aw-pl, 2025-03-25:
@filipswit Regarding AD5259 - did you check max power consumption from the B-W pins? Max is 20mA

@gkasprow :

> @filipswit  Since we have IN_N in FMC we delete it on AFE, right? And what with 4 signals IN_N[0..3] leave them uncconected on connectors or tie to GND?
Do we disconnect now those signals **IN_N[0..3]**  from the FMC board and tight also to GND?

aw-pl, 2025-03-25:
@filipswit You are using Vlogic 3.3V on 2.5V I2C bus @ AD5259  . Did you check if voltage levels are OK? Or maybe connect to P2V5?

filipswit, 2025-03-25:
> @gkasprow :
> 
> @filipswit Since we have IN_N in FMC we delete it on AFE, right? And what with 4 signals IN_N[0..3] leave them uncconected on connectors or tie to GND?
> Do we disconnect now those signals IN_N[0..3] from the FMC board and tight also to GND?

The idea of leaving unconnected IN_N[0..3] signals on FMC is to keep compatibllity with AFE v1.0 to not shortcut IN_N[0..3] to GND on FMC, but I'm not sure if this scenario is useful.

> @filipswit You are using Vlogic 3.3V on 2.5V I2C bus @ AD5259 . Did you check if voltage levels are OK? Or maybe connect to P2V5?

For both AD5259 and MAX11617, supply voltage is from 2.7V, so we have to use 3.3V. High level is recognized form 0.7xVCC=2.31V, so it should be fine to use 2.5V I2C bus.

filipswit, 2025-03-29:
> @filipswit Regarding AD5259 - did you check max power consumption from the B-W pins? Max is 20mA

B-W pins are loaded with 100k, so the max current is about 30uA

aw-pl, 2025-04-01:
@filipswit **WARNING**
I think that I2C lines are **swapped**. Please double check that and confirm:

[image]

and PCB:

[image]

schematic:

[image]

aw-pl, 2025-04-01:
@filipswit **WARNING** We are discussing the AD5259 - need to check the proper resistance, since this chip have a big production spread +/- 30%. We are planning to add in series 15kOhm 0.1% to compensate the values spread and still keep up with good step resolution of the regulation. We need also to check LP2998 regarding values spreads on internal divider, if this is also provided.

aw-pl, 2025-04-01:
We are discussing the LP2998  - since there is a chapter 9.2.4 about level shifting for the same purpose as we want. The difference however is here that this configuration shown in the chapter uses external resistors, so this is without the influence of internal LP2998 resistors (50k/50k) that can have big value spread (still keeping proportions) like in case of AD5229. So the fig 27 configuration could be better.

filipswit, 2025-04-01:
> @filipswit WARNING
> I think that I2C lines are swapped. Please double check that and confirm:

Good catch! There is a bug in library! Fixed on schematics

[image]

We can do a circuit as in datashet: 

[image]

Output voltage will be in range 1,09V to 1,571V (in worst case, when digital trim pot is -30% tolerance)
Biggest current over dig trim pot is when BW=0R, U=3,15-1,09=2,06 -->I=U/R=2,06/430R=4,7mA

filipswit, 2025-04-03:
After some discusion with @aw-pl, to eliminate +/- 30% toleralne of dig trim pot and increase the resolution. I propose this connection with variant to revert primary solution.

[image]

filipswit, 2025-04-04:
@gkasprow I place simple 5k voltage divider at VDDQ input, which has internal 100k divider.

[image]

aw-pl, 2025-04-05:
@filipswit In my opinion for alternative mounting variant you should connect: W to SEN, and A to VTT. Please double check that.

filipswit, 2025-04-06:
Agree that's safer solution.