elhep/FMC_ADC_128CH_10B #10: Voltages monitoring

aw-pl, 2025-03-20:
> @gkasprow I think we were also talking to add some simple ADC for monitoring power supply voltages for FMC board (including IN_N readout). Do you confirm to also add this? 

 _Originally posted by @aw-pl in [#5](https://github.com/elhep/FMC_ADC_128CH_10B/issues/5#issuecomment-2738558456)_

- [ ] Monitor voltages from FMC connector: 
- P3V3**_FMC**
- P12V0 <---- this could be difficult to do, so maybe without it?
- V_ADJ
- [ ] Monitor local power supplies: from _Supply_FMC_ADC128_: 
- 2.1V
- 3.3V
- 1.9V
- 3.15V
- 1.2V
- 2.5V
- 1.8V
- 5.0V_A
- [ ] Extras:
- PGOOD signal
- IN_N <- still question from @filipswit (#5) do we keep those signals on the connector?

This would give in total 11-12 voltages to monitor. 
For example we could use such chip on I2C in small package:
https://www.analog.com/en/products/max11616.html with 12 channels
?
**Note**: This uses clock streaching (question about current i2c implementations) -> however using EXT_CLK mode is possible to avoid this.

aw-pl, 2025-03-20:
or ADS7952 for example

filipswit, 2025-03-20:
How about 4ch ADS1115IDGST, I can try to fit 3.

aw-pl, 2025-03-21:
In general I would go for MAX11616 (or reduced in number of channels version). This ADS* were somewhere used as you say, but it is limited in functionality. Especially it cannot do continues scans over channels automatically, as well you need to check for conversion done (pooling register) before readout <- meaning a lot of communication. Where MAX is more automated, for example just gives you two times 0's on readout due to conversion latency and then concatenates the result, so it is done already during readout.

gkasprow, 2025-03-21:
maybe also monitor AFE voltages ?

filipswit, 2025-03-21:
So we have to use MAX11617EEE+T to cover all rails. I'll try to fit it, but it is not small..

gkasprow, 2025-03-21:
This functionality changes the BOM significantlym @aw-pl did you discuss it with pcb manufacturer? Do we have time to add extra features?

filipswit, 2025-03-22:
I was thinking about it, but we can't use MAX11613 x3 because they have only one I2C address.

filipswit, 2025-03-22:
After some movements on PCB it fits.

I think I've placed all interesting rails:

[image]

aw-pl, 2025-03-24:
@gkasprow I didn't discuss this with the manufacturer. Do we revert this change due to BOM? 
@filipswit What would be time estimation to add this feature for you?

filipswit, 2025-03-24:
I've already routed MAX11617 on PCB.

filipswit, 2025-03-29:
Changed MAX11617 supply from P3V3 to P3V3_FMC

aw-pl, 2025-04-01:
Checked schematic, pinouts and PCB layout of chip as discussed