elhep/FMC_ADC_128CH_10B #8: Add power LEDs with labels

aw-pl, 2025-03-20:
Add LEDs for power supply indication with labels (best to be at front of FMC if easy to route):

- [ ] P12V0
- [ ] P3V3_FMC
- [ ] P1V8A
- [ ] P2V5A

filipswit, 2025-03-20:
Can I place LEDs on TOP layer? Do you use panel in FMC module?

[image]

gkasprow, 2025-03-21:
Doesnt matter until it does not collide with cable

filipswit, 2025-03-23:
Added LEDs, finally I squizzed them on bottom, so it will be possoble to use with FMC panel, I've also changed LD1 to the same as added LEDs - KPA-1606CGCK.

aw-pl, 2025-03-24:
Check - did you also added labels on PCB?

filipswit, 2025-03-29:
Yes:

[image]