elhep/FMC_ADC_128CH_10B #9: CAL_ON signal

aw-pl, 2025-03-20:
Use the free port on FMC from #3 to send CAL_ON signal to AFE board (to be validated with AFE issues)

aw-pl, 2025-03-24:
I was checking and we can make **TX_TRIG_SEL** constant.
What I suggest to make this selectable by resistor (1 or 0).
The **default** setting would be **0/low**, as I'm using it now.

This signal we could then use for calibration - **CAL_ON**
Check if the level translator is necessary (however we have a free slot after PDN_GBL

filipswit, 2025-03-27:
If there will be enough space to fit I2C expander, I will connect **CAL_ON** to FPGA, and **TRIG_SEL** and **PDN_GBL** to expander. Is it good idea?

aw-pl, 2025-03-28:
That would be good - only to check if default state of **PDN_GBL** is low at power-up of expander. Otherwise we don't need that much this connection.

filipswit, 2025-03-29:
- Connected **CAL_ON** to FMC - **HB20_N** and to J1C pin 146
- Added I2C expander TCA9536 for **TX_TRIG_SEL**, default state of expander is high, so I don't connect **PDN_GBL** - leave it just with pull-up

[image]

[image]

aw-pl, 2025-04-01:
@filipswit Checked, questions:

- [ ] Is internal weak pull-up OK for IC20?
- [ ] Pleasy clarify: `I don't connect PDN_GBL - leave it just with pull-up` -> you mean with **PULL_DOWN**?

filipswit, 2025-04-01:
> Is internal weak pull-up OK for IC20?

IC20 on start up is configured as input with 100k pull-up and IC1 has 71k pull-down, so the state will be undefined, but when you will configure IC20 as output it will be push-pull.

> Pleasy clarify: I don't connect PDN_GBL - leave it just with pull-up -> you mean with PULL_DOWN?

yes ofcourse, pull-down

aw-pl, 2025-04-04:
**WARNING** TX_TRIG_SEL should be connected to IC6 instead of IC1, due to direction.

filipswit, 2025-04-04:
Fixed:

[image]

aw-pl, 2025-04-05:
OK, checked