elhep/FMC_ADC_128CH_10B #13: FFT measurements

aw-pl, 2025-04-03:
FMC ADC:
on FMC ADC we are having some harmonics in range of 500kHz to 4MHz on -55dB level

[image]

gkasprow, 2025-04-03:
did you enable to 20MHz filter?

aw-pl, 2025-04-03:
Yes, it is with the filter -> 200mV BW 20MHz

aw-pl, 2025-04-04:
In fact the previous FFT was due that the measurement cable was very close to AFCKU power supply section, that is generating this spectra.
Moving away from this field results in much better FFT :)

[image]

aw-pl, 2025-04-04:
As a note, here is a spectra on SDA or SCL (similar results) test points near to Si570 when there is no transmission of data.

[image]